microsoft/DMF #13: Why is the Virtual Hid Device KMDF-only?

hansmbakker, 2018-09-02:
The Virtual Hid Device module is KMDF-only:

https://github.com/Microsoft/DMF/blob/8d126adc31c7c9d925dce7caa8ca016290babe79/Dmf/Modules.Library/Dmf_VirtualHidDeviceVhf.h#L24

However, the [vhidmini2 sample is also supported in UMDF](https://github.com/Microsoft/Windows-driver-samples/tree/master/hid/vhidmini2/driver/umdf2).

Is there a reason for which the Virtual Hid module is implemented as KMDF-only?

hansmbakker, 2018-09-02:
I now see that a HID minidriver (in the referenced sample) is not the same as using the Virtual Hid Framework.

hansmbakker, 2018-09-04:
I believe the PortableDeviceButtons module uses VirtualHidDevice, but that is KMDF-only, too, since it depends on VirtualHidDevice.

Eventually I’d like to create a solution to [this](https://stackoverflow.com/q/45773666/1114918), which I’m really struggling with, and I had hoped that DMF would eventually make it less difficult.

samtertzakian, 2018-09-04:
I remember why DMF does not support VHF in User-mode: In Vhf.h this code exits:

#ifndef _KERNEL_MODE
#error User Mode Support is currently not implemented
#endif

The underlying DDK does not support VHF in User-mode so DMF will not either. If that support is added in the future, then DMF will do so also. 

Most Modules work for both Kernel and User-modes. The goal is that Modules should be agnostic to whether they run in Kernel or User-mode (when it is possible).

As for an example that uses DMF_VirtualHidDeviceVhf see DMF_HidPortableDeviceButtons which is in in the Library right now.

hansmbakker, 2018-09-04:
> I remember why DMF does not support VHF in User-mode: In Vhf.h this code exits:

Yes, that’s what I meant with
> I now see that a HID minidriver (in the referenced sample) is not the same as using the Virtual Hid Framework, and that the Virtual Hid Framework does not work in UMDF.

> see DMF_HidPortableDeviceButtons

That’s what I meant with
> I believe the PortableDeviceButtons module uses VirtualHidDevice, but that is KMDF-only, too, since it depends on VirtualHidDevice.

Do you know whether there are any plans to add a HID minidriver module to provide a UMDF solution?

samtertzakian, 2018-09-04:
Three things...

1. For your purpose, I *think* you should use Dmf_HidTarget instead of Dmf_VirtualHidDeviceVhf. Dmf_HidTarget will allow you to talk to a specific HID device. Dmf_VirtualHidDeviceVhf is designed so that you can create virtual HID devices.)
2. I *think* we are releasing a BLE related Module in the near future...this week or next. (Someone is working on that and the plan was to release it, but I am not 100% sure.) That Module may help you.
3. Dmf_HidPortableDeviceButtons is built on Dmf_VirtualHidDeviceVhf.

Re:
Do you know whether there are any plans to add a HID minidriver module to provide a UMDF solution?

It sounds like you are asking if VHF will support User-mode eventually. I do not know the answer to this.

Disclaimer: All my comments are my own opinion and not official Microsoft statements. This posting is provided "AS I" with no warranties, and confers no right.

hansmbakker, 2018-09-05:
@samtertzakian thanks for thinking along!
1. I'm not looking to talk to a HID device but I'm trying to emulate a HID device - I'm looking for a way to emulate a Surface Dial's HID events based on incoming BLE GATT messages, because I cannot modify the firmware of the device that I have.
2. That sounds really interesting! I understand you can't make a promise to it.
3. I got that, thanks :)

About a HID minidriver - as I understood it, a HID minidriver is the older, but UMDF-supported way to emulate a HID device. VHF is KMDF-only indeed. I also would prefer VHF to support UMDF.

My issue with KMDF drivers is that you cannot develop, run and distribute them as a private person (because you'll need a corporate developer account to get Microsoft to sign it, otherwise it won't work when Secure Boot is enabled).

samtertzakian, 2018-09-11:
Original question for this issue has been answered above (VHF is only supported in Kernel-mode by DDK) so this issue will be closed.

closed 2018-09-11 by samtertzakian