microsoft/DMF #53: Code analyzer produces warning on DMF_MODULE_DECLARE_CONTEXT

laengand, 2019-03-25:
Hi
When i running the code analyzer with Minimum Driver Rules. I get a C6031 warning. The warning is related to `DMF_MODULE_LIVEKERNELDUMP_POINTER_STORE`.
```
#define DMF_MODULE_DECLARE_CONTEXT(ModuleName)                                                           \
WDF_DECLARE_CONTEXT_TYPE_WITH_NAME(DMF_CONTEXT_##ModuleName##,                                           \
                                   ##ModuleName##ContextGet);                                            \
VOID                                                                                                     \
DMF_##ModuleName##_LiveKernelDumpInitialize(_In_ DMFMODULE DmfModule)                                    \
{                                                                                                        \
    DMF_CONTEXT_##ModuleName##* moduleContext;                                                           \
                                                                                                         \
    moduleContext = ##ModuleName##ContextGet(DmfModule);                                                 \
    DMF_MODULE_LIVEKERNELDUMP_POINTER_STORE(DmfModule,moduleContext,sizeof(DMF_CONTEXT_##ModuleName##)); \
}                                                                                                        \
```
I see that the `#define` evaulates to `DMF_LiveKernelDump_DataBufferSourceAdd` which have the `_Must_inspect_result_` annotation. `DMF_##ModuleName##_LiveKernelDumpInitialize` does not inspect the return value. So I would expect that either `DMF_LiveKernelDump_DataBufferSourceAdd` should have the `_Must_inspect_result_` removed or `DMF_##ModuleName##_LiveKernelDumpInitialize` should inspect the return value. 
Any thoughts on this ?

samtertzakian, 2019-03-25:
Hi, we are merging a new sample this week. I will try to have a resolution for this issue, one way or another, in that merge. Thank you for bringing it up.

samtertzakian, 2019-03-25:
Because that Module instatiates the Ring Buffer using "ringBufferModuleConfig.Mode = RingBuffer_Mode_DeleteOldestIfFullOnWrite;" it means it writing to RingBuffer never fails since it is never full. It means we can simply remove "must inspect result". That will be merged this week with new sample.
Thank you again. 
It is strange we did not notice this before...Let me investigate why that is.

laengand, 2019-03-26:
Hi 
Good to hear. Nice and fast feedback. Keep up the good work :-)

closed 2019-03-26 by laengand

samtertzakian, 2019-03-26:
The change is submitted in to master now. LKG v1.1.7.