microsoft/DMF #62: DMF in Visual Studio WDK templates?

hansmbakker, 2019-05-22:
This is not a bug - I just wanted to say that it would be great if the WDK templates in Visual Studio would make use of DMF.

Not sure whether this issue is the right place to share this feedback though.

samtertzakian, 2019-05-22:
Yes, it is in the plan to make a DMF Driver Wizard or template of some kind. But we are currently working on writing unit tests against the Modules to stress test the Framework and Modules. 

If you are interested in doing that work, we welcome it. 

Thank you for the suggestion.

hansmbakker, 2019-05-24:
I submitted a feature request at https://developercommunity.visualstudio.com/content/problem/579179/it-would-be-great-if-the-wdk-templates-in-visual-s.html.

I'm quite new to Windows drivers so I feel I'm not the right person for this, but great that it is part of the plan.

AraHaan, 2019-07-07:
Is there any way to manually create a project file for a software driver I want to write when my device does not have internet to make the wdf installer work?

hansmbakker, 2019-07-07:
@AraHaan I believe your question is not related to the question of this github issue. Also I am not sure it is DMF specific. Could you please delete your question from this issue and open a new issue for it?

AraHaan, 2019-07-07:
I hope I created the issue in the right place. But at least it was filed in a Microsoft repository.

nefarius, 2022-02-10:
Not a 100% what you're after but I created [a little helper](https://github.com/nefarius/WDF-Utils/tree/master/DMF) to make DMF inclusion a breeze!

paulpv, 2024-11-13:
> Not a 100% what you're after but I created [a little helper](https://github.com/nefarius/WDF-Utils/tree/master/DMF) to make DMF inclusion a breeze!

Thanks @nefarius!
I'm new to DMF so just found this.
Glad to see that your helper got merged into this code per https://github.com/Microsoft/DMF/blob/master/Dmf/Documentation/Driver%20Module%20Framework.md#simplifying-compilation-and-linking-with-dmf.
I use it and it helped.

I still ran into compiler errors and created a new issue https://github.com/microsoft/DMF/issues/260 that I think there is another snippet of code needed in the project file in order for a developer writing their own DMF driver to successfully compile.
```xml
  <ItemGroup Label="WrappedTaskItems">
    <ClCompile Include="DmfInterface.c">
      <WppEnabled>true</WppEnabled>
      <WppKernelMode>true</WppKernelMode>
      <WppTraceFunction>TraceEvents(LEVEL,FLAGS,MSG,...)</WppTraceFunction>
      <WppGenerateUsingTemplateFile>{km-WdfDefault.tpl}*.tmh</WppGenerateUsingTemplateFile>
    </ClCompile>
  </ItemGroup>
```