microsoft/DMF #67: Framework package to link against?

hansmbakker, 2019-06-03:
(Coming from .Net background, not sure if this is common for driver development)

Seeing that C++/WinRT is distributed as a NuGet package now, I'm wondering whether DMF could be released as a (NuGet) package as well?

Meaning - a packaged version of a compiled version together with the required headers?

Advantages for developers would be a easier consumption, a cleaner project repository and clear dependency management.

samtertzakian, 2019-06-04:
We will discuss internally and get back to you. Thank you for your suggestion. (Internally, we use Nuget packages for our own private DMF in our build system.)

hansmbakker, 2019-07-15:
I was wondering whether you’ve had the chance of discussing this, and what the outcome was?

(I can imagine that even if your colleagues approved it could still require effort since currently it is used for internal builds that probably need to stay separated?)

hansmbakker, 2019-08-30:
Hi Sam,
could you say something about the state of this issue?

samtertzakian, 2019-09-23:
Hi, Hans,

We have not had a chance to address this issue directly. However, it is on our list. We have been using Nuget internally with DMF and other libraries. *There will not be any progress on this issue from our end for at least another couple of months due to our schedules.* For this reason, this issue remains open.

We have discussed this issue and we plan to work on it. Our solution needs to work seamlessly in our organization which is very large and has many consumers including our build organization.

hansmbakker, 2019-09-23:
Thank you for making this clear!

samtertzakian, 2020-01-15:
Hi, Hans. Internally we are reorganizing how we use DMF from Github. So we addressing the issues you raised in this thread. I am not sure how long it will take or what the final outcome will be. I will keep you up to date here however. We have not forgotten this issues.

samtertzakian, 2020-09-25:
Hi, Hans, we are still working on this task but making progress. Just fyi...

nefarius, 2022-02-10:
Not a 100% what you're after but I created [a little helper](https://github.com/nefarius/WDF-Utils/tree/master/DMF) to make DMF inclusion a breeze!

samtertzakian, 2022-02-12:
Hi nefarius, 

Do you think we should include your file in the Dmf directory and and update the documentation to include what you wrote?
We could update all the sample drivers to use the .props file.

If you would like to create a PR with that it would be great. Or, I can do it.

I need to try it out myself, but I think what you did is simpler than the legacy manner in which the samples include DMF now.

nefarius, 2022-02-12:
> Do you think we should include your file in the Dmf directory and and update the documentation to include what you wrote?
> We could update all the sample drivers to use the .props file.

Sure, if it makes life easier for everyone! 

> If you would like to create a PR with that it would be great. Or, I can do it.
> 
> I need to try it out myself, but I think what you did is simpler than the legacy manner in which the samples include DMF now.

I'll try to migrate the samples in my fork and when successful will gladly provide a PR. Whoever makes it first 😉

samtertzakian, 2022-02-15:
Let put the Dmf.Props file next to the DmfVersion.h file. Then, lets update the documentation "Special Topics" section with the text you wrote. Does that sound good? I can do it if you like. I just don't want to do it if you started it.

nefarius, 2022-02-15:
Sounds good! No I haven't started any work on this yet.

samtertzakian, 2022-03-30:
There is a PR pending to resolve this issue: https://github.com/microsoft/DMF/pull/210

Samples have been updated to illustrate the use of this method. Documentation has also been updated. Thank you, nefarius for the suggestion. 

Release v1.1.119 will have this change.

samtertzakian, 2022-03-30:
Master has the update. Next month it will be in Release v1.1.19.

closed 2022-03-30 by samtertzakian