microsoft/DMF #90: Suggestion: move Modules.Template to samples

hansmbakker, 2019-12-27:
The code in `Modules.Template` seems to be used for the samples mostly (besides serving as a template for new modules).

Currently, the code in `DmfSamples` / in `DmfSamples.sln` seems to contain wrapper code mostly, while it seems not to be needed for the `lib` code that is generated by `Dmf.sln`. Also, when looking through the samples, you need to look into two solutions to have an overview of the complete driver.

Would it be an idea to move this code to the DmfSamples folder and solution, so that the Dmf folder & solution only contain the "library code" and so that there is a better separation between library & samples?

samtertzakian, 2019-12-30:
The purpose of DMF_Template.c is two-fold: 1. Provide a template for a new Module. 2. Provide function signatures for the different callbacks. 

The Samples solution and projects only contain Client drivers that use DMF. Those drivers use either DMF Library directly, or use the DMF Template Library (and the DMF Library indirectly).  The Sample solution does not contain any Modules or Module Libraries. Modules and Module Libraries should go in the DMF Solution. 

People what want their own Module Libraries generally will create a sperate DMF Solution that contains just their own private  Modules and Module Libraries they write. And separately, they will have Client drivers that link to those Libraries (and DMF Libraries indirectly). 

Another option is that they can have a single solution that contains their Module Library as well as their own Client drivers that link to their private Module Libraries.

Here is how DMF looks as distributed:

"DMF Core"
       "DMF Library"
              "DMF Template Library"

Client can write a new Module Library (DMF Private Library) that links to "DMF Library" in the same way the "DMF Template Library" links to "DMF Library".

The Vendor (you)  should not add "DMF Private Library" to DMF Solution because when we update DMF the Client would need to update DMF Solution every time. For this reason it is best for the Vendor to create a separate solution for "DMF Private Library". The Vendor can also add the Vendor's Client drivers to that solution.

Let me know if that is not clear.

hansmbakker, 2019-12-30:
Hi Sam,
my point was just that I was wondering whether the `Modules.Template` project should be in the `Dmf.sln` solution rather than in a separate solution or in the `DmfSamples.sln` solution.

The purpose of the various parts is clear to me.

Personally I created a private library in a separate solution together with the client driver, that works well for me.

samtertzakian, 2020-01-08:
Hi, Hans,
From our point of view, it is best to have Template in the DMF solution since the Template project shows how to write Modules/Module Libraries that are similar to the other projects. The Samples solution uses the Template project. I think we can clarify this by doing either of two things:
1. Renaming "Template" to "Templates and Samples" 
2. Making another Library that just has "Samples" so we would have "Templates" and "Samples".
I think moving Template to Sample drivers, however, is not a good option.
Let me mull it over with coworkers and get back to you.
Thank you for your suggestion, however.

closed 2020-01-08 by samtertzakian