microsoft/DMF #92: Unused inc folder

hansmbakker, 2019-12-29:
A lot of the samples add the `inc` folder as additional includes folder, while for most samples this folder does not exist.

Examples:
https://github.com/microsoft/DMF/blob/465ecb1953eea0024c98b821b280a6eed2f05223/DmfSamples/VHidMini2Dmf/User/VHidMini2DmfU.vcxproj#L103

https://github.com/microsoft/DMF/blob/master/DmfSamples/SwitchBar4/SwitchBar4.vcxproj#L133

samtertzakian, 2019-12-29:
I am planning on cleaning up the samples' projects in the next few days.
When I do I will check that. However all samples should compile correctly
without any change as I do that almost every day. If not, let me know.

Also, please see the slide deck i added recently.  It has some slides
devoted to steps to building Modules. Look at "Part 2".

hansmbakker, 2019-12-30:
Agree that they compile fine, I just noticed it and decided to share my findings.

samtertzakian, 2019-12-30:
Thank you for you notes, Hans. Look for an update this week or early next week due to holidays.

hansmbakker, 2019-12-30:
I understand, enjoy the New Year celebration! Happy New Year :)

samtertzakian, 2020-01-10:
Fixed in Release 1.1.39. Thank you for your feedback.

closed 2020-01-10 by samtertzakian