microsoft/DMF #93: DMF_GENERIC_xxx callbacks might not be needed

hansmbakker, 2019-12-30:
In
https://github.com/microsoft/DMF/blob/465ecb1953eea0024c98b821b280a6eed2f05223/Dmf/Modules.Template/Dmf_Template.c#L1035-L1036
there is written that you need to set all unused callbacks to a generic callback, but in the other module implementations it looks like you can just remove those lines of code that use a generic implementation?

samtertzakian, 2019-12-30:
Hi, Hans,

What that text means is: Use "Search/Replace" to replace the string "_Template_" with the name of the new Module you are writing for all the callbacks you will use. You do not need to use all the callbacks...In most cases you will use just a few callbacks.

"_GENERIC_" is probably not a good name for the example. It just means the name of your new Module.

Recently we uploaded a PowerPoint presentation that discusses many aspects of DMF. Please see slide #81 which describes the steps to Create a Module. They are probably better than the steps described in the text you are using. I am going to update the comments in the Dmf_Template.c file to make the steps more clear. I will try to do so this week with the next merge.

You can look at many of the Modules and see that most Modules use just a few callbacks.

If this response does not answer your question, let me know.

hansmbakker, 2019-12-30:
> You can look at many of the Modules and see that most Modules use just a few callbacks.

Yes, I understood that, it was just that I wanted to give feedback on the comment in `Dmf_Template.c`.

samtertzakian, 2019-12-31:
Hi, Hans,
In LKG 1.1.37 (updated just now) I have updated Dmf_Template.c's comments to accurately reflect the current best practices. Thank you for indicating that issue. The new comments are much better and contain a reference to the PowerPoint slides.

closed 2019-12-31 by samtertzakian