microsoft/DMF #94: Where best to ask for help?

hansmbakker, 2019-12-30:
(Sorry if this is the wrong place)

I saw the dmf-feedback@microsoft.com mail address, there is the Issues section of this Github repository and there is the [driver development & WDK section in the MSDN forums](https://social.msdn.microsoft.com/Forums/vstudio/en-US/home?forum=wdk).

When I [search for "DMF"](https://social.msdn.microsoft.com/Forums/vstudio/en-US/home?category=&forum=wdk&filter=&sort=relevancedesc&brandIgnore=true&searchTerm=dmf) in the driver development / WDK forums, I get zero hits. That suggests that there are few people with DMF knowledge in the forums, and I do not want to make a mess of the Issues section of this repository.

What would be the best place to ask for help / troubleshooting regarding DMF driver development?

hansmbakker, 2019-12-30:
(The concrete issue I have is that my driver works partially, but that I cannot open a connection to it to send ioctls to it - neither using a win32 CLI app nor using [UWP with a `CustomCapability`](https://docs.microsoft.com/en-us/windows-hardware/drivers/devapps/hardware-support-app--hsa--steps-for-driver-developers).
The error I get is `HRESULT: 0x8007001F` / `A device attached to the system is not functioning`. 
`DeviceWatcher` correctly finds my driver device but the error is thrown when I call `CustomDevice.FromIdAsync(Id, DeviceAccessMode.ReadWrite, DeviceSharingMode.Exclusive)` or `CreateFile(...)` in the win32 app.)

samtertzakian, 2019-12-30:
Hi, Hans,

For now, I think asking questions here is fine. But, if you want to send a question privately, please send it to dmf-feedback@microsoft.com. That alias reaches several people including me. (If this ever becomes an issue, I will let you know.)

I can tell you that we have successfully used DMF with "DeviceWatcher". The first thing I would to diagnose your issue is see if you can access the IOCTLs using a native Win32 application. If that does not work, then nothing else will work. If that does work, then we know there is another issue. One way to debug the issue since you say you can see the device from the app but not send IOCTLs is to put a break point at this function: DmfContainerEvtDeviceIoControl(). That should get called when you send an ICOTL. If that gets called, then you can step through the code until it gets to your Module or find out why it does not get there.  One more thing make sure you are not using "Internal" DeviceIoControl in the Module as that will only work for driver to driver communication.

Do not hesitate to contact privately at dmf-feedback@microsoft.com and one of us will help you if we can. All the feedback we receive helps us to fix issues we did not see in our own usage of DMF.

hansmbakker, 2019-12-31:
Thank you very much, Sam! I sent a mail to dmf-feedback@microsoft.com containing all details, a bug reproduction branch, test apps, repro steps.

> The first thing I would to diagnose your issue is see if you can access the IOCTLs using a native Win32 application
The issue is that this does not work, even though I defined my IOCTLs and registered them using the `Dmf_IoctlHandler` module. There is a test app in the bug reproduction branch for this.

If any details are missing, please let me know.

samtertzakian, 2020-01-01:
Hi, Hans,
I have received your email. I will close this issue after a resolution is achieved. I will also post the resolution here prior to closing (in addition to responding to your email of course).

samtertzakian, 2020-01-08:
The issue you encountered was because you are trying to send IOCTls to a device interface generated by a filter driver. This, however, is not allowed by underlying WDM. More information is here:
https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/using-control-device-objects
The solution is to use a Control Device (not possible in UMDF) or piggy back the communication using INPUT/OUTPUT/FEATURE reports.

closed 2020-01-08 by samtertzakian