microsoft/DMF #96: Why are lib files added to linker manually?

hansmbakker, 2020-01-02:
In the Dmf projects I see that the includes and libraries are added manually in the project settings using "Additional Include directories" and "Additional dependencies".

Coming from the C# world, I'm used to the "Add reference" feature for linking projects together which makes Visual Studio manage the references between projects (it automatically includes the necessary DLLs files without you worrying about the paths). Also, this lets Visual Studio understand the dependencies between projects, so that Visual Studio will (re)build the required projects in the correct order.

Say, e.g. that you would change something in `DmfUModules.Library`, which would be referenced by `DmfUModules.Template`, and you recompile `DmfUModules.Template`, then I believe it will see that it also needs to build `DmfUModules.Library` for the changes to come through.

According to https://stackoverflow.com/a/29579275/1114918 this is also possible for C/C++ (both static libraries and DLLs). Is there a reason for which, in DMF, adding the dependencies manually is more practical? Or could this make the project setup easier?

samtertzakian, 2020-01-13:
Hi, Hans,

Let me explain how linking DMF works and best practices the current time. Keep in mind that we use DMF internally ourselves.

In the DMF solution, you will see the "DmfK" and "DmfU" projects. Those projects are just libraries (.lib). If you look at the "Librarian" tab in properties you can see the DmfK contains these libraries: DmfKFramework.lib;DmfKModules.Library.lib;DmfKModules.Library.Tests.lib.

When a Client driver needs to link to DMF, it only needs to link to DmfK.lib. Doing that will allow it to access all the code in: DmfKFramework.lib;DmfKModules.Library.lib;DmfKModules.Library.Tests.lib.

Now, DMF is designed so that programmers should create their own Modules. Those Modules should not be added to the above projects because they are common for all users. Instead, programmers should create their own Module Library (eg. DmfModules.Private.lib)  and link their driver to BOTH DmfK.lib and DmfModules.Private.lib.

DmfKModules.Library.Template is an example (and template) for such a private Module Library. The DMF Sample drivers link to both DmfK.lib and DmfK.Modules.Template.lib because we use Template as a repository for all samples. (I know that name should be changed to Templates and Samples).

So, you should create your own library and link to both your own library as well as DmfK.lib.

(Same applies for User-mode, but use "U" instead.) 

One more thing, much to my dismay the #pragma comment directive seems to not work when building drivers...at least did not work a couple years ago.

Does that answer your question? If not, let me know here or using the email alias.

hansmbakker, 2020-01-15:
Hi Sam,
thank you for commenting.

My question was not per say about the dependency structure (what depends on what) but more about *how* the dependencies are *implemented*.

Currently the Librarian / Linker input is put manually in the Project properties and I was wondering whether that could be replaced by using the `project reference` functionality as explained in [this stackoverflow answer](https://stackoverflow.com/a/29579275/1114918) and as shown below:
[image]

I think `#pragma` is a different way of doing things?

samtertzakian, 2020-01-16:
I see, Hans. Let me try that ns see if it works. I will get back to you and I will re-open the issue. Thank you for the clarification. I agree that if the above works, it is more intuitive.

hansmbakker, 2020-01-16:
Personally I couldn’t get it to work (I got a reference error but that might be a bug in my preview version of Visual Studio), but according to the stackoverflow article I linked it should be possible.

Thank you for looking into it!

samtertzakian, 2022-03-30:
This issue is resolved by: https://github.com/microsoft/DMF/pull/210

Simply add a copy of DMF.props file to the Client Driver Visual Studio Project file and drivers will compile and link automatically without explicitly setting paths and names of .lib files.

This method works without any UI by simply editing the project files.

closed 2022-03-30 by samtertzakian