microsoft/DMF #135: Isn't it better to design DMF on C++ code base?

moolwalk, 2020-08-31:
Inspired by the sample driver source code of [UcmUcsiAcpiSample](https://github.com/microsoft/Windows-driver-samples/tree/master/usb/UcmUcsiAcpiSample/UcmUcsiAcpiSample), designing DMF on C++ code base will be much efficient project and help KMDF driver developers to be more productive.

samtertzakian, 2020-09-25:
Hi, thank you for your feedback. DMF (like WDF) is usable by both C and C++ drivers. Although C++ has some nice features that would have helped us, doing so would have forced all developers to use C++ for their drivers. DMF is designed so that it can be added to existing drivers as well as new drivers. The general idea behind DMF is that Modules are object-oriented pieces of code that can be used together to create more complex Modules so the important ideas behind C++ are implemented.

Sorry for the delay in answering this question. Please contact dmf-feedback@microsoft.com if you have other questions or would like to discuss this issue furthere.

moolwalk, 2020-09-25:
After studying DMF more, I figured that a C++ DMF wrapper can be made to encapsulate it to meet my request. I've started to experiment with that approach. Thank you.

closed 2020-09-25 by moolwalk

samtertzakian, 2020-09-27:
Yes, encapsulation by C++ classes has been discussed over the years. It would be great to have that as an option as some people have expressed interest in that. If you do that work please place those files in a separate directory so that we can easily merge the changes internally. Clients would include the C++ headers from the new directories and all the support would be there and reference the Modules in the existing directories. Also, when you that work, remember that these DMF supports both Kernel and User-modes so that may affect your work if you decided to support both modes. Thank you for your interest. I look forward to seeing that.