microsoft/DMF #150: Unexpected behaviour of ScheduledTask module

nefarius, 2021-01-26:
Greetings!

First of all; this might be on me for misconfiguring or misusing this module so let me quickly summarize what I try to achieve. I have a device which requires a buffer sent to periodically while it's powered on, for example like every 10ms and/or sooner on demand.

Therefore I chose the `ScheduledTask` module which offers a deferred and "retry on success" pattern. This is unfortunately not working for me (the callback as seen below is only fired once after invoked), here's my setup:

```c
DMF_MODULE_ATTRIBUTES moduleAttributes;
DMF_CONFIG_ScheduledTask dmfSchedulerCfg;

// ... other logic ...

DMF_CONFIG_ScheduledTask_AND_ATTRIBUTES_INIT(
&dmfSchedulerCfg,
&moduleAttributes
);

dmfSchedulerCfg.EvtScheduledTaskCallback = DMF_OutputReportScheduledTaskCallback;
dmfSchedulerCfg.CallbackContext = deviceContext;
dmfSchedulerCfg.PersistenceType = ScheduledTask_Persistence_NotPersistentAcrossReboots;
dmfSchedulerCfg.ExecutionMode = ScheduledTask_ExecutionMode_Deferred;
dmfSchedulerCfg.ExecuteWhen = ScheduledTask_ExecuteWhen_Other;
dmfSchedulerCfg.TimerPeriodMsOnSuccess = 10;
dmfSchedulerCfg.TimerPeriodMsOnFail = 10;

DMF_DmfModuleAdd(
DmfModuleInit,
&moduleAttributes,
WDF_NO_OBJECT_ATTRIBUTES,
&deviceContext->OutputReportScheduler
);
```

Here's my callback:

```c
ScheduledTask_Result_Type
DMF_OutputReportScheduledTaskCallback(
	_In_ DMFMODULE DmfModule,
	_In_ VOID* CallbackContext,
	_In_ WDF_POWER_DEVICE_STATE PreviousState)
{
	// ... send buffer to device ...
	
	return ScheduledTask_WorkResult_SuccessButTryAgain;
}
```

I invoke it like so (which works, but only **once** until called again):

```c
DMF_ScheduledTask_ExecuteNowDeferred(
		Context->OutputReportScheduler,
		Context
	);
```

Before I do a deep-dive into dissecting the module; any obvious error here in my thinking of what the module can do for me or the configuration I posted? Thanks!

samtertzakian, 2021-02-05:
Sorry for delayed response.

If I recall, you need to use ScheduledTask_ExecutionMode_Deferred in order for ScheduledTask_WorkResult_SuccessButTryAgain to function the way you expect.

Let me verify this and get back to you after I verify. I know we are using it as a periodic timer in the way you expect.

samtertzakian, 2021-02-05:
Ok, I am verifying now... but I think I know what the issue is:

You need to use this call to start the scheduled task:

NTSTATUS
DMF_ScheduledTask_ExecuteNowDeferred(
    _In_ DMFMODULE DmfModule,
    _In_opt_ VOID* CallbackContext
    );

You cannot use:

ScheduledTask_Result_Type
DMF_ScheduledTask_ExecuteNow(
    _In_ DMFMODULE DmfModule,
    _In_ VOID* CallbackContext
    );

nefarius, 2021-02-07:
Thank you very much for the insights, I will test with `ScheduledTask_ExecuteWhen_D0Entry` then, should be alright.

Appreciate the addition to documentation!

One more question though; am I then by design allowed to still call `DMF_ScheduledTask_ExecuteNowDeferred()` "in between" when I need to not await the specified timeout value? I'd like to utilize it that way to send a packet to my device ASAP when necessary and sticking to just using the scheduler callback would allow me to not care about synchronization as the only sending code would remain in that callback.

Cheers!

samtertzakian, 2021-02-10:
Yes, you instantiate using `ScheduledTask_ExecuteWhen_D0Entry`. Then, you can use `DMF_ScheduledTask_ExecuteNow()` to cause the same callback to be immediately called. In this case, however, you may need to acquire and release a lock in the callback to make sure that the deferred callback using timer does not race with your immediate execution. In fact, there is a comment in the code that explains that. If I recall, we specifically added `DMF_ScheduledTask_ExecuteNow()` for the exact purpose you describe.

Having said all that, I will update the documentation to explain all this (and possibly add a Method to do what you originally intended).

nefarius, 2021-02-24:
> Yes, you instantiate using `ScheduledTask_ExecuteWhen_D0Entry`. Then, you can use `DMF_ScheduledTask_ExecuteNow()` to cause the same callback to be immediately called. In this case, however, you may [...]

It works now exactly as I need it with these adjustments, thanks! Also introduced a lock just to be safe.

samtertzakian, 2021-07-21:
The pending PR has a fix for this issue: https://github.com/microsoft/DMF/pull/178

samtertzakian, 2021-07-27:
Resolved by Release v1.1.98. Please use new DMF_ScheduledTask_ExecuteNowDeferredEx() Method. It honors the return value correctly.

closed 2021-07-27 by samtertzakian