microsoft/DMF #166: Latest code generates following linker errors in my project

alkovacs, 2021-06-09:
I pulled the latest DMF code and now I am getting the following linker errors in my driver project...

1>user_code.obj : error LNK2019: unresolved external symbol _invalid_parameter_noinfo referenced in function memcpy_s
1>user_code.obj : error LNK2019: unresolved external symbol _errno referenced in function memcpy_s
1>DmfK.lib(DmfUtility.obj) : error LNK2019: unresolved external symbol __stdio_common_vswprintf referenced in function _vsnwprintf_l
1>DmfK.lib(DmfCore.obj) : error LNK2019: unresolved external symbol __stdio_common_vsprintf referenced in function _vsnprintf_l
1>DmfK.lib(Dmf_CrashDump.obj) : error LNK2019: unresolved external symbol __stdio_common_vsprintf_s referenced in function _vsprintf_s_l

samtertzakian, 2021-06-09:
Hi

What version of the compiler are you using?
Are you using EWDK?

Let me investigate this and get back to you later today.

alkovacs, 2021-06-09:
Microsoft Visual Studio Professional 2019
Version 16.10.0
VisualStudio.16.Release/16.10.0+31321.278
Microsoft .NET Framework
Version 4.8.04084

Installed Version: Professional
Visual C++ 2019   00435-60000-00000-AA343
Microsoft Visual C++ 2019
ASP.NET and Web Tools 2019   16.10.525.31942
ASP.NET and Web Tools 2019
Cookiecutter   16.10.21111.19
Provides tools for finding, instantiating and customizing templates in cookiecutter format.
Debugging Tools for Windows   10.0.20324.1000
Integrates the Windows Debugger functionality (http://go.microsoft.com/fwlink/?linkid=223405) in Visual Studio.
IntelliCode Extension   1.0
IntelliCode Visual Studio Extension Detailed Info
Microsoft JVM Debugger   1.0
Provides support for connecting the Visual Studio debugger to JDWP compatible Java Virtual Machines
Microsoft MI-Based Debugger   1.0
Provides support for connecting Visual Studio to MI compatible debuggers
Microsoft Visual C++ Wizards   1.0
Microsoft Visual C++ Wizards
Microsoft Visual Studio VC Package   1.0
Microsoft Visual Studio VC Package
NuGet Package Manager   5.10.0
NuGet Package Manager in Visual Studio. For more information about NuGet, visit https://docs.nuget.org/
ProjectServicesPackage Extension   1.0
ProjectServicesPackage Visual Studio Extension Detailed Info
Test Adapter for Boost.Test   1.0
Enables Visual Studio's testing tools with unit tests written for Boost.Test.  The use terms and Third Party Notices are available in the extension installation directory.
Test Adapter for Google Test   1.0
Enables Visual Studio's testing tools with unit tests written for Google Test.  The use terms and Third Party Notices are available in the extension installation directory.
Visual Studio Code Debug Adapter Host Package   1.0
Interop layer for hosting Visual Studio Code debug adapters in Visual Studio
Visual Studio Tools for CMake   1.0
Visual Studio Tools for CMake
Windows Driver Kit   10.0.20324.1000
Headers, libraries, and tools needed to develop, debug, and test Windows drivers (msdn.microsoft.com/en-us/windows/hardware/gg487428.aspx)

samtertzakian, 2021-06-09:
Hi, so...

There have been issues reported with 16.10. The issues I found are different than what you found above. However, it is possible that you have run in to a similar issue. 

The issue that I have found has just been fixed yesterday and release was issued here:

https://docs.microsoft.com/en-us/visualstudio/releases/2019/release-notes

You can see June 8 release for 16.10.1. Can I trouble you to install that version of Visual Studio and see if it corrects your issue?

This is the issue I found that has been corrected: 

https://developercommunity.visualstudio.com/t/compiler-regression-in-16100-with-ewdk/1435981?from=email

I have just built all DMF, DmtTest and DmfSamples without any problem.

If you still have a problem, let me know. You can also send me email at: dmf-feedback@microsoft.com

Thank you for your report and your patience.

alkovacs, 2021-06-09:
So I think I see what's going on, I updated to 2019 V16.0.1 as you suggested, tried the builds (DMF Debug and My Project Debug) again and my project build failed. I was using the 10.0.20344.0 SDK/DDK . Switched that to 10.0.19041.0 SDK/DDK and my project build failed with, not the same but similar errors. Switched to the 10.0.17763.0 SDK/DDK and everything seems to compile/link fine. BUT, I went back to the 10.0.20344.0 SDK/DDK and the 10.0.19041.0 SDK/DDK, in both cases my project build fails. Then upon switching back to the 0.0.17763.0 SDK/DDK , my project build fails again with just 2 errors (linker) like **__stdio_common_vsprintf referenced in function _vsnprintf_l** .... Seems something is not being cleaned out in the settings ???

samtertzakian, 2021-06-09:
Are you using the SDK/DDK or the EWDK? We always use EWDK here.

We open the EWDK command using command line as Administrator. Then, run LaunchBuildEnv.cmd. Then, switch to BuildEnv and run:

setdevenv
setupbuildenv
setupvsenv

Then, we run VisualStudio from that command line...using a shortcut: "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\IDE\devenv.exe"

I have not used standard SDK/DDK in years. The EWDK has worked very well.

Can you try that?

It looks to me like you are running into some kind of SDK/DDK issue. I think you will be happier with EWDK. One nice thing about it is that you can easily have multiple versions of EWDK installed and then you can start VS in any version you want.

alkovacs, 2021-06-10:
So after installing the EWDK and following the instructions at the installation link, using VS 2019 16.0.1 , we were able to build the DMF (Debug) and our Project (Debug) in SDK/DDK version 10.0.19041.0 which is what was in the 
[Download the EWDK for Windows 10, version 2004](url) . Funny now my post driver build analysis tool no longer runs. So there is ONLY 1 SDK/WDK version available in an EWDK? I did see ddk's for win 7 and 8 , but none of the newer ones, other than 10.0.19041.0. Is it possible to have multiple under on EWDK install?

samtertzakian, 2021-06-10:
> Download the EWDK for Windows 10, version 2004 <http://url> . Funny now

Right, there is only a single SDK/DDK in the EWDK. That is why it works so
well.

But there are multiple EWDK packages. You just open the command line in the
directory where the package resides (it can be mounted as CD if you want).

I am not sure if the older EWDK packages are still available.  Let me look
and get back to you.

samtertzakian, 2021-06-10:
Some other information about EWDK:

https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/using-the-enterprise-wdk

alkovacs, 2021-06-11:
Sam, thank you for all of your help in what turned **out NOT to be an issue specifically with the DMF**, but an environment issue...

closed 2021-06-11 by alkovacs

samtertzakian, 2021-06-11:
Thank you for your patience and for reporting the issue.

soubhikSM, 2021-06-14:
Just to add here, I was facing same issue in my driver project. Without making any changes to the SDK and WDK, downgrading from 16.10.1 VS'19 to 16.9.6 helped solve the issue and build was successful.

samtertzakian, 2021-06-14:
Hi, soubhikSM,

Do you mean you downgraded from 16.10.1 to 16.9.6? Or from 16.10.0.

16.10.1 is what corrects some known issues (ones I reported), but I am not sure if other issues exist.

soubhikSM, 2021-06-15:
Hey samtertzakian , yes I downgraded from 16.10.1 to 16.9.6.

I and a couple of my other teammates faced this same issue with 16.10.1.

<filename>.obj : error LNK2019: unresolved external symbol _invalid_parameter_noinfo referenced in function memcpy_s
<filename>.obj : error LNK2019: unresolved external symbol _errno referenced in function memcpy_s
<filename>.obj : error LNK2019: unresolved external symbol __stdio_common_vsnwprintf_s referenced in function _vsnwprintf_s_l
<filename2>.obj : error LNK2019: unresolved external symbol __stdio_common_vswprintf referenced in function _vsnwprintf_l
E:\<path>\driver\x64\Debug\<filename>.sys : fatal error LNK1120: 4 unresolved externals
Done building project "<filename>.driver.vcxproj" -- FAILED.

I went through the comments here and finally after having tried lot of other stuff, installing enterprise version 16.9.6 resolved the issue. I did not need to install EWDK.

samtertzakian, 2021-06-15:
Hi, soubhikSM,

I left feedback to Visual Studio via the link in upper right corner.
That is how I was able to get a fix for 16.10.0 to 16.10.1 (which corrected the issue in EWDK building).
Apparently, there are other issues.

I will just send them the actual errors and see what they say.

Sorry, there is not much else I can do directly.

samtertzakian, 2021-06-16:
Hi

I have just been informed that a fix for these above linker errors is in the Visual Studio Release pipeline. I don't know when it will be released or in which version.

But for now, apparently the work-around is:

Clear the "External Include Directories" field in the project settings.

Or, just use 16.9.x until then.

soubhikSM, 2021-06-16:
Thanks for the update samtertzakian. :)