microsoft/DMF #194: Unused config param in DMF_VirtualHidKeyboard module

serkodev, 2021-11-27:
These config param in DMF_VirtualHidKeyboard module are not been used:
- `HidDescriptorLength`
- `HidDescriptor`

Related file:
https://github.com/microsoft/DMF/blob/v1.1.108/Dmf/Modules.Library/Dmf_VirtualHidDeviceVhf.h
https://github.com/microsoft/DMF/blob/v1.1.108/Dmf/Modules.Library/Dmf_VirtualHidDeviceVhf.c
https://github.com/microsoft/DMF/blob/v1.1.108/Dmf/Modules.Library/Dmf_VirtualHidDeviceVhf.md

Is there any purpose for keeping it or just placeholder for deprecated config?
I can help to submit a pull request for cleaning the code (including the related sample) if it is really unused code.

samtertzakian, 2021-11-28:
It appears this is a holdover from the original version of virtual HID support in DMF when it used the DMF_VirtualHidMini Module (the legacy method before VHF support).

From what I can see, VHF does not need that...But now I am wondering how IOCTL_HID_GET_DEVICE_DESCRIPTOR is supported in VHF,

You can see how it was used previously in DMF_VirtualHidMini#1208. Let me verify tomorrow, but I think you can go ahead and create a PR. 

Note: Some people are able to create PRs here and others are not. Please try it and let me know.

samtertzakian, 2021-12-01:
Hi, we have corrected this issue internally and we will merge it here this week. So, please don't make a PR to save your time. Thank you for the feedback.

closed 2021-12-01 by samtertzakian

serkodev, 2021-12-01:
Cool, thanks for working on it. Sorry for the late response of my personal work.

samtertzakian, 2021-12-01:
No need to apologize. I think we will make an update by end of week.