microsoft/DMF #195: How can I disable WPP for the DMF driver ?

serkodev, 2021-11-27:
Currently WPP is force to enable WPP for tracing, however I just want to use `DbgPrint` instead of it.
Is it possible to disable WPP when developing driver in DMF?

samtertzakian, 2021-11-28:
Hi, Let me look into this...I think it is possible by making a small change in DMF Trace file...I think...not sure. But, if you want to do that so you can get running logging in Windbg this is how you do it using DmfKTest as an example.

!wmitrace.stop DmfKTest -kd
!wmitrace.start DmfKTest -kd
!wmitrace.enable DmfKTest {61C379CE-3A6B-4E34-B8B1-BEF18A0F6209} -level 0x7 -flag 0xFFFF
!wmitrace.dynamicprint 1

The GUID is the driver's tracing GUID as seen here: 
https://github.com/microsoft/DMF/blob/master/DmfTest/DmfKTest/sys/Trace.h

If you do that...it is the equivalent of DbgPrint(). Having said that...let me see if it is possible to do what you asked.

samtertzakian, 2021-11-30:
Hi...

I have made branch here that "converts" WPP into DbgPrint: https://github.com/microsoft/DMF/tree/DbgPrint
You can see that doing so is non-trivial. One of the big issues is that WPP tracing supports things like %!STATUS!. 
I have placed this branch here for now in case you absolutely need it and to show the complications involved. This code is not well tested and should only be used for debug purposes. It is unlikely we will ever integrate this code into master branch however. There is also a change in the project file so it is not possible to easily switch from DbgPrint() to WPP.

My feeling is that you want to use DbgPrint() so that you can see logging while your driver is running. You might want that because you don't realize that !wmitrace in Windbg supports this. Instead of using this branch, please try what I wrote above:

!wmitrace.stop DmfKTest -kd
!wmitrace.start DmfKTest -kd
!wmitrace.enable DmfKTest {61C379CE-3A6B-4E34-B8B1-BEF18A0F6209} -level 0x7 -flag 0xFFFF
!wmitrace.dynamicprint 1

I hope this helps you...Let me know if !wmitrace solves your issue. If so, we can just delete this branch. If not, let me know why and we can try to resolve that.

serkodev, 2021-12-01:
Thanks for your fast response and make a sample. I will try asap.

My purpose is just want to have a disable WPP optional because WPP is not necessary for developing a driver.
Without WPP, the binary will be more lightweight and less dependency (WppRecorder.sys). Also the setup of the project would be simpler.

samtertzakian, 2021-12-01:
Oh, you want to simply disable WPP entirely...not route to DbgPrint(). Let me think about that.

samtertzakian, 2021-12-02:
Ok, I looked into this a bit. I don't see a way to actually remove the WPP statements from being compiled. It is possible to do what I did before which is route them to DbgPrint() as I did above and make DbgPrint not emit anything. However, the strings will still be in the code. The calls will still be made which is what you are trying to avoid. 

My advice is to simply leave the WPP tracing in DMF as it is. Microsoft is using WPP in all its drivers as well as for IFR tracing (internal buffers). The additional overhead of space and time (both minimal) is low. 

Note that with WPP the strings are not actually in the code...they are stripped out of the driver binary and only present in the driver's .pdb file. We have not seen or had any complaints related to WPP tracing causing performance issues (memory or time).

I will close this issue as: Can't (and won't) do.

closed 2021-12-02 by samtertzakian