microsoft/DMF #216: EyeGazeIoctl sample fails ApiValidator

lablabla, 2022-06-23:
I'm trying to test the Eye gaze driver from the samples. I've taken the sample as is, compiled it on Windows 10 64bit.

It failed the first test of `ApiValidator` with
`EyeGazeIoctl.sys has unsupported API call to "ntoskrnl.exe!KeGetCurrentIrql"`

I've looked at the `Gaze HID` sample project and the provided sample driver from [here](https://github.com/MSREnable/GazeHid#sample-driver), and running the tests on this driver passes the `ApiValidator`
However this file was updated 2 weeks ago.

samtertzakian, 2022-06-24:
We will investigate this issue. Thank you for reporting it.

samtertzakian, 2022-11-03:
Hi 

Sorry for the delay in responding to this issue...I have no repro for this issue:

````
C:\EWDK>apivalidator -DriverPackagePath:C:\Users\x\DmfSamples\EyeGazeIoctl\x64\Debug\ -SupportedApiXmlFiles:"C:\EWDK\EWDK_co_release_22000_210604-1628\Program Files\Windows Kits\10\build\universalDDIs\x64\universalddis.xml"
ApiValidation: Calling aitStatic 'aitstatic.exe'.
ApiValidation:
Summary:
        "C:\Users\satertza\source\Repos\Microsoft\DMF0\DmfSamples\EyeGazeIoctl\x64\Debug" is Universal

ApiValidation: All binaries are Universal
````

I receive no error either via visual Studio (2019) nor by running the test explicitly as I did above.

If you still have this issue, please show me how you are reproducing it.

closed 2022-11-03 by samtertzakian