microsoft/DMF #224: Building with Windows 11, version 22H2 WDK (10.0.22621.0) supported?

nefarius, 2022-10-18:
I prepared a fresh Windows 11 machine with latest Visual Studio 2022 and SDK/WDK Windows 11, version 22H2 WDK (10.0.22621.0) but attempting to build DMF fails with the following errors:

```text
Build started...
1>------ Build started: Project: DmfUModules.Library, Configuration: Debug x64 ------
2>------ Build started: Project: DmfUFramework, Configuration: Debug x64 ------
3>------ Build started: Project: DmfKFramework, Configuration: Debug x64 ------
4>------ Build started: Project: DmfKModules.Library, Configuration: Debug x64 ------
2>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(314,5): error MSB4044: The "ValidateNTTargetVersion" task was not given a value for the required parameter "ValidNTTargetVersions".
2>Done building project "DmfUFramework.vcxproj" -- FAILED.
1>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(314,5): error MSB4044: The "ValidateNTTargetVersion" task was not given a value for the required parameter "ValidNTTargetVersions".
1>Done building project "DmfUModules.Library.vcxproj" -- FAILED.
5>------ Build started: Project: DmfUModules.Library.Tests, Configuration: Debug x64 ------
6>------ Build started: Project: DmfUModules.Template, Configuration: Debug x64 ------
5>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(314,5): error MSB4044: The "ValidateNTTargetVersion" task was not given a value for the required parameter "ValidNTTargetVersions".
5>Done building project "DmfUModules.Library.Tests.vcxproj" -- FAILED.
6>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(314,5): error MSB4044: The "ValidateNTTargetVersion" task was not given a value for the required parameter "ValidNTTargetVersions".
6>Done building project "DmfUModules.Template.vcxproj" -- FAILED.
7>------ Build started: Project: DmfU, Configuration: Debug x64 ------
7>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(314,5): error MSB4044: The "ValidateNTTargetVersion" task was not given a value for the required parameter "ValidNTTargetVersions".
7>Done building project "DmfU.vcxproj" -- FAILED.
3>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(314,5): error MSB4044: The "ValidateNTTargetVersion" task was not given a value for the required parameter "ValidNTTargetVersions".
3>Done building project "DmfKFramework.vcxproj" -- FAILED.
4>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(314,5): error MSB4044: The "ValidateNTTargetVersion" task was not given a value for the required parameter "ValidNTTargetVersions".
4>Done building project "DmfKModules.Library.vcxproj" -- FAILED.
8>------ Build started: Project: DmfKModules.Library.Tests, Configuration: Debug x64 ------
9>------ Build started: Project: DmfKModules.Template, Configuration: Debug x64 ------
8>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(314,5): error MSB4044: The "ValidateNTTargetVersion" task was not given a value for the required parameter "ValidNTTargetVersions".
8>Done building project "DmfKModules.Library.Tests.vcxproj" -- FAILED.
9>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(314,5): error MSB4044: The "ValidateNTTargetVersion" task was not given a value for the required parameter "ValidNTTargetVersions".
9>Done building project "DmfKModules.Template.vcxproj" -- FAILED.
10>------ Build started: Project: DmfK, Configuration: Debug x64 ------
10>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(314,5): error MSB4044: The "ValidateNTTargetVersion" task was not given a value for the required parameter "ValidNTTargetVersions".
10>Done building project "DmfK.vcxproj" -- FAILED.
========== Build: 0 succeeded, 10 failed, 0 up-to-date, 0 skipped ==========
```

As an experiment I tried to set the Target OS Version to Windows 10 or newer but that blew up even more:

[image]

nefarius, 2022-10-18:
I've advanced a tiny bit by following [this workaround](https://gitlab.kitware.com/cmake/cmake/-/issues/23717) but am still unable to build the UMDF projects:

```text
Build started...
1>------ Build started: Project: DmfUModules.Library, Configuration: Debug x64 ------
2>------ Build started: Project: DmfUFramework, Configuration: Debug x64 ------
1>Building 'DmfUModules.Library' with toolset 'WindowsUserModeDriver10.0' and the 'Desktop' target platform.
1>Dmf_ActivitySensor.cpp
1>Dmf_BufferPool.c
1>Dmf_BufferQueue.c
1>Dmf_DeviceInterfaceMultipleTarget.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_BufferPool.c)
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_BufferQueue.c)
1>Dmf_Doorbell.c
1>Dmf_GpioTarget.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_DeviceInterfaceMultipleTarget.c)
1>Dmf_HashTable.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_ActivitySensor.cpp)
1>Dmf_IoctlHandler.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_HashTable.c)
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_Doorbell.c)
1>Dmf_RingBuffer.c
1>Dmf_SpiTarget.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_IoctlHandler.c)
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_GpioTarget.c)
1>Dmf_Stack.c
1>Dmf_String.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_RingBuffer.c)
1>Dmf_AlertableSleep.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_SpiTarget.c)
1>Dmf_CmApi.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_Stack.c)
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_String.c)
1>Dmf_ComponentFirmwareUpdate.c
1>Dmf_ComponentFirmwareUpdateHidTransport.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_CmApi.c)
1>Dmf_DefaultTarget.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_AlertableSleep.c)
1>Dmf_File.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_ComponentFirmwareUpdateHidTransport.c)
1>Dmf_HingeAngle.cpp
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_HingeAngle.cpp)
1>Dmf_I2cTarget.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_ComponentFirmwareUpdate.c)
1>Dmf_Interface_BusTarget.c
1>C:\Users\BenjaminHöglinger-St\Documents\DMF\Dmf\Framework\DmfIncludes_KERNEL_MODE.h(152,10): fatal  error C1083: Cannot open include file: 'ntifs.h': No such file or directory (compiling source file ..\..\Modules.Library\Dmf_DefaultTarget.c)
1>Dmf_Interface_ComponentFirmwareUpdate.c
[... cut ...]
5>------ Build started: Project: DmfU, Configuration: Debug x64 ------
5>Building 'DmfU' with toolset 'WindowsUserModeDriver10.0' and the 'Desktop' target platform.
5>C:\Program Files\Microsoft Visual Studio\2022\Professional\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(1542,5): error MSB6006: "Lib.exe" exited with code 1181.
5>LINK : fatal error LNK1181: cannot open input file 'DmfUFramework.lib'
5>Done building project "DmfU.vcxproj" -- FAILED.
========== Build: 0 succeeded, 5 failed, 0 up-to-date, 0 skipped ==========
```

nefarius, 2022-10-18:
Alright, got UMDF projects to build by specifying my minimum UMDF versions in the projects files and rebuilding again, here's my diff:

```diff
 Dmf/Solution/DmfU/DmfU.vcxproj                     | 24 +++++++++++++++++++
 Dmf/Solution/DmfUFramework/DmfUFramework.vcxproj   | 12 ++++++++++
 .../DmfUModules.Library.Tests.vcxproj              | 12 ++++++++++
 .../DmfUModules.Library.vcxproj                    | 12 ++++++++++
 .../DmfUModules.Library.vcxproj.filters            | 10 ++++----
 .../DmfUModules.Template.vcxproj                   | 28 ++++++++++++++++++++--
 6 files changed, 91 insertions(+), 7 deletions(-)

diff --git a/Dmf/Solution/DmfU/DmfU.vcxproj b/Dmf/Solution/DmfU/DmfU.vcxproj
index 9b9dd48..6c75426 100644
--- a/Dmf/Solution/DmfU/DmfU.vcxproj
+++ b/Dmf/Solution/DmfU/DmfU.vcxproj
@@ -48,6 +48,30 @@
     <UseDebugLibraries Condition="'$(Configuration)'=='Debug'">True</UseDebugLibraries>
     <WindowsTargetPlatformVersion>$(LatestTargetPlatformVersion)</WindowsTargetPlatformVersion>
   </PropertyGroup>
+  <PropertyGroup Label="Configuration" Condition="'$(Configuration)|$(Platform)'=='Debug|ARM64'">
+    <UMDF_VERSION_MAJOR>2</UMDF_VERSION_MAJOR>
+    <UMDF_VERSION_MINOR>25</UMDF_VERSION_MINOR>
+  </PropertyGroup>
+  <PropertyGroup Label="Configuration" Condition="'$(Configuration)|$(Platform)'=='Release|ARM64'">
+    <UMDF_VERSION_MAJOR>2</UMDF_VERSION_MAJOR>
+    <UMDF_VERSION_MINOR>25</UMDF_VERSION_MINOR>
+  </PropertyGroup>
+  <PropertyGroup Label="Configuration" Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">
+    <UMDF_VERSION_MAJOR>2</UMDF_VERSION_MAJOR>
+    <UMDF_VERSION_MINOR>25</UMDF_VERSION_MINOR>
+  </PropertyGroup>
+  <PropertyGroup Label="Configuration" Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">
+    <UMDF_VERSION_MAJOR>2</UMDF_VERSION_MAJOR>
+    <UMDF_VERSION_MINOR>25</UMDF_VERSION_MINOR>
+  </PropertyGroup>
+  <PropertyGroup Label="Configuration" Condition="'$(Configuration)|$(Platform)'=='Debug|x64'">
+    <UMDF_VERSION_MAJOR>2</UMDF_VERSION_MAJOR>
+    <UMDF_VERSION_MINOR>25</UMDF_VERSION_MINOR>
+  </PropertyGroup>
+  <PropertyGroup Label="Configuration" Condition="'$(Configuration)|$(Platform)'=='Release|x64'">
+    <UMDF_VERSION_MAJOR>2</UMDF_VERSION_MAJOR>
+    <UMDF_VERSION_MINOR>25</UMDF_VERSION_MINOR>
+  </PropertyGroup>
   <PropertyGroup>
     <DebuggerFlavor>DbgengKernelDebugger</DebuggerFlavor>
     <SigningCertPath Condition="'$(SigningCertPath)'==''">$(SigningCertificatesDir)OEMTest_OS_DRIVER.pfx</SigningCertPath>
diff --git a/Dmf/Solution/DmfUFramework/DmfUFramework.vcxproj b/Dmf/Solution/DmfUFramework/DmfUFramework.vcxproj
index 7ba02d9..51e0dad 100644
--- a/Dmf/Solution/DmfUFramework/DmfUFramework.vcxproj
+++ b/Dmf/Solution/DmfUFramework/DmfUFramework.vcxproj
@@ -46,31 +46,43 @@
     <UseDebugLibraries>true</UseDebugLibraries>
     <PlatformToolset>WindowsUserModeDriver10.0</PlatformToolset>
     <WindowsTargetPlatformVersion>$(LatestTargetPlatformVersion)</WindowsTargetPlatformVersion>
+    <UMDF_VERSION_MAJOR>2</UMDF_VERSION_MAJOR>
+    <UMDF_VERSION_MINOR>25</UMDF_VERSION_MINOR>
   </PropertyGroup>
   <PropertyGroup Condition="'$(Configuration)|$(Platform)'=='Release|Win32'" Label="Configuration">
     <UseDebugLibraries>false</UseDebugLibraries>
     <PlatformToolset>WindowsUserModeDriver10.0</PlatformToolset>
     <WindowsTargetPlatformVersion>$(LatestTargetPlatformVersion)</WindowsTargetPlatformVersion>
+    <UMDF_VERSION_MAJOR>2</UMDF_VERSION_MAJOR>
+    <UMDF_VERSION_MINOR>25</UMDF_VERSION_MINOR>
   </PropertyGroup>
   <PropertyGroup Condition="'$(Configuration)|$(Platform)'=='Debug|x64'" Label="Configuration">
     <UseDebugLibraries>true</UseDebugLibraries>
     <PlatformToolset>WindowsUserModeDriver10.0</PlatformToolset>
     <WindowsTargetPlatformVersion>$(LatestTargetPlatformVersion)</WindowsTargetPlatformVersion>
+    <UMDF_VERSION_MAJOR>2</UMDF_VERSION_MAJOR>
+    <UMDF_VERSION_MINOR>25</UMDF_VERSION_MINOR>
[... 179 lines cut ...]
+  </PropertyGroup>
   <Import Project="$(VCTargetsPath)\Microsoft.Cpp.props" />
   <PropertyGroup>
     <OutDir>$(SolutionDir)$(ConfigurationName)\$(PlatformName)\individual_libs\$(ProjectName)\</OutDir>
@@ -86,7 +110,7 @@
       <LanguageStandard>stdcpp17</LanguageStandard>
     </ClCompile>
     <ClCompile Condition="'$(Platform)'=='ARM64'">
-      <AdditionalOptions Condition="'$(_NT_TARGET_VERSION)'=='$(_NT_TARGET_VERSION_WIN10_CO)'" >/d2guardsignret %(AdditionalOptions)</AdditionalOptions>
+      <AdditionalOptions Condition="'$(_NT_TARGET_VERSION)'=='$(_NT_TARGET_VERSION_WIN10_CO)'">/d2guardsignret %(AdditionalOptions)</AdditionalOptions>
     </ClCompile>
     <Link>
       <AdditionalDependencies>%(AdditionalDependencies);hid.lib;setupapi.lib;cfgmgr32.lib;</AdditionalDependencies>
@@ -94,7 +118,7 @@
   </ItemDefinitionGroup>
   <ItemDefinitionGroup Condition="'$(Platform)'=='ARM64'">
     <Link>
-      <AdditionalOptions Condition="'$(_NT_TARGET_VERSION)'=='$(_NT_TARGET_VERSION_WIN10_CO)'" >/guard:delayloadsignret %(AdditionalOptions)</AdditionalOptions>
+      <AdditionalOptions Condition="'$(_NT_TARGET_VERSION)'=='$(_NT_TARGET_VERSION_WIN10_CO)'">/guard:delayloadsignret %(AdditionalOptions)</AdditionalOptions>
     </Link>
   </ItemDefinitionGroup>
   <ItemGroup>
```

I doubt this is the way to go but at least I can now build my updated driver again 😅 (driver in question uses IDDCX so I have to obey its minimum UMDF version requirements).

samtertzakian, 2022-10-18:
Have you tried building using the EWDK? That is how we build here.
However, we have not yet tried Visual Studio 2022 due to other issues with our internal build system.
To build with EWDK download it and mount it. Then, open a command line window and properly initialize using a couple of commands listed on page where you download. Then, start Visual Studio from inside the command window. That may work. I will try that and let you know if it works without modifications. Or if you try it, let me know.

samtertzakian, 2022-10-19:
I have verified that building DMF with 22H2 WDK fails. We are investigating. I will update this thread when i know more. We are discussing with EWDK team.

nefarius, 2022-10-19:
Thanks for the feedback, I'll pause my tests then for now and keep building and testing my driver with the patch until we discover the "proper" way.

An interesting bit; if you open the file `C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.Default.props` and uncomment line 27:

```xml
	<!-- This is commented as setting it here over writes the one in the drivergenral.xml and it shows Windows10 rather than the value in xml file --> 
    <TargetVersion Condition="'$(TargetVersion)' == ''">$(LatestTargetVersion)</TargetVersion>
```

The troubles with other projects (kernel, non-DMF etc.) go away. But I assume that comment is there for a reason so I'd like to see this investigated properly as well 👍 

Cheers

nefarius, 2022-10-20:
One more discovery, it appears that Win32/x86 now refuses to build altogether. Has 32-Bit support quietly been dropped from the WDK? I hope not, I dislike having to support it still but taking away that choice should not be up to MS...

samtertzakian, 2022-10-20:
I have not heard that x86 support is removed. I will find out soon once we get a fix from WDK/VS team. I know they are actively looking into this issue. I will ask about x86 when I get a chance.

samtertzakian, 2022-11-03:
Release v.1.1.129 corrects this issue. The user-mode settings have been updated to indicate the version of UMDF to use which causes build environment to use UMDF. However, it is also necessary to run this command before starting Visual Studio 2022:

`set MSBuildUserExtensionsPath=%VSINSTALLDIR%\MsBuild`

It is anticipated that later versions of the EWDK will not need the extra command.

This release works with EWDK_ni_release_svc_prod1_22…621_220804-1759.iso as well as the previous EWDK.

closed 2022-11-03 by samtertzakian