microsoft/DMF #232: fatal error C1510: Cannot load language resource clui.dll

77, 2023-02-28:
Fresh clone of master. Command line .\build\build.ps1 20H1 Release ARM64.

77, 2023-02-28:
``PS C:\data\src\DMF> .\build\build.ps1 20H1 Release ARM64

Major                           : 1
Minor                           : 1
Patch                           : 131
PreReleaseTag                   :
PreReleaseTagWithDash           :
PreReleaseLabel                 :
PreReleaseLabelWithDash         :
PreReleaseNumber                :
WeightedPreReleaseNumber        : 60000
BuildMetaData                   :
BuildMetaDataPadded             :
FullBuildMetaData               : Branch.master.Sha.7d60fe2b875eaef27f51bad02bd336c3eefcb653
MajorMinorPatch                 : 1.1.131
SemVer                          : 1.1.131
LegacySemVer                    : 1.1.131
LegacySemVerPadded              : 1.1.131
AssemblySemVer                  : 1.1.131.0
AssemblySemFileVer              : 1.1.131.0
FullSemVer                      : 1.1.131
InformationalVersion            : 1.1.131+Branch.master.Sha.7d60fe2b875eaef27f51bad02bd336c3eefcb653
BranchName                      : master
EscapedBranchName               : master
Sha                             : 7d60fe2b875eaef27f51bad02bd336c3eefcb653
ShortSha                        : 7d60fe2
NuGetVersionV2                  : 1.1.131
NuGetVersion                    : 1.1.131
NuGetPreReleaseTagV2            :
NuGetPreReleaseTag              :
VersionSourceSha                : 7d60fe2b875eaef27f51bad02bd336c3eefcb653
CommitsSinceVersionSource       : 0
CommitsSinceVersionSourcePadded : 0000
UncommittedChanges              : 0
CommitDate                      : 2022-11-18

##vso[build.updatebuildnumber]1.1.131
Mounting ISO: C:\data\src\DMF\.cache\20H1.iso
Enabling EWK from D
**********************************************************************
** Enterprise Windows Driver Kit (WDK) build environment
** Version vb_release.19041.1
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.0
** Copyright (c) 2019 Microsoft Corporation
**********************************************************************
Microsoft (R) Build Engine version 16.3.0+0f4c62fea for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

Building the projects in this solution one at a time. To enable parallel build, please add the "-m" switch.
Build started 2/28/2023 10:38:02 AM.
Project "C:\data\src\DMF\DMF.sln" on node 1 (default targets).
ValidateSolutionConfiguration:
  Building solution configuration "Release|ARM64".
Project "C:\data\src\DMF\DMF.sln" (1) is building "C:\data\src\DMF\Dmf\Solution\DmfK\DmfK.vcxproj.metaproj" (2) on node
 1 (default targets).
Project "C:\data\src\DMF\Dmf\Solution\DmfK\DmfK.vcxproj.metaproj" (2) is building "C:\data\src\DMF\Dmf\Solution\DmfKMod
ules.Template\DmfKModules.Template.vcxproj.metaproj" (3) on node 1 (default targets).
Project "C:\data\src\DMF\Dmf\Solution\DmfKModules.Template\DmfKModules.Template.vcxproj.metaproj" (3) is building "C:\d
ata\src\DMF\Dmf\Solution\DmfKModules.Library\DmfKModules.Library.vcxproj" (4) on node 1 (default targets).
DriverBuildNotifications:
  Building 'DmfKModules.Library' with toolset 'WindowsKernelModeDriver10.0' and the 'Desktop' target platform.
  Using KMDF 1.15.
PrepareForBuild:
  Creating directory "C:\data\src\DMF\Release\ARM64\individual_libs\DmfKModules.Library\".
  Creating directory "ARM64\Release\DmfKModu.7CD49BFE.tlog\".
InitializeBuildStatus:
  Creating "ARM64\Release\DmfKModu.7CD49BFE.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
RunWpp:
  D:\Program Files\Windows Kits\10\\bin\10.0.19041.0\\x86\tracewpp.exe -ext:.c.C.cpp.CPP.h.H -cfgdir:"D:\Program Files\
[... 214 lines cut ...]
  D:\Program Files\Microsoft Visual Studio\2019\BuildTools\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(487,5)
: error MSB6006: "CL.exe" exited with code 4. [C:\data\src\DMF\Dmf\Solution\DmfUFramework\DmfUFramework.vcxproj]

    0 Warning(s)
    3 Error(s)

Time Elapsed 00:00:06.55
Unmounting C:\data\src\DMF\.cache\20H1.iso

Attached          : False
BlockSize         : 0
DevicePath        :
FileSize          : 10766188544
ImagePath         : C:\data\src\DMF\.cache\20H1.iso
LogicalSectorSize : 2048
Number            :
Size              : 10766188544
StorageType       : 1
PSComputerName    :

Exception: C:\data\src\DMF\build\build.ps1:224
Line |
 224 |              throw "Build failed (exitCode=${exitCode})"
     |              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     | Build failed (exitCode=1)

samtertzakian, 2023-02-28:
Hi

It looks to me like you are using an older EWDK. From log it seems to be:

** Enterprise Windows Driver Kit (WDK) build environment
** Version vb_release.19041.1

Is it possible you can use a later EWDK and see if it fixes the issue.

Here we are using "Windows 11, version 22H2" with VS 2019. It also works with VS 2022, I believe.

If that works, it is it possible you can use that new EWDK?

If not, we may have to make changes to the project files to support the older EWDK.

If i get some time, I can try to reproduce your issue using the older EWDK and see how difficult it is to fix that.

77, 2023-03-01:
The 19041.1 EWDK is the latest version supported by the build.ps1 script. I did try building in VS2022 with the 22H2 WDK but got a different error with that. I can build ARM64 in VS2022 with the 22H2 WDK but x64 and Win32 builds give another error. That was why I tried using the build.ps1 script from the repo.

samtertzakian, 2023-03-01:
Can you try the following...

Use your desired EWK. Open a command line prompt and run:

cd EWDK_co_release_22000_210604-1628\       (use your desired version)
LaunchBuildEnv.cmd
cd BuildEnv
SetDevEnv.cmd
SetupBuildEnv.cmd
SetupVSEnv.cmd

Then, run visual studio from the command line prompt and open the DMF solution.
Rebuild All and everything should build.

As for the build.ps1, I have not used that myself...It is pretty old and perhaps it is not working with current projects. I will try to investigate why. However, the above instructions should work for you.

77, 2023-03-02:
Using the Vb EWDK I already had downloaded and issuing the above commands following by `msbuild Dmf.sln /p:Configuration=Debug /p:Platform=x64` gives the same error. 

Launching VS2022 from that command prompt and building gives the following error.

`10>d:\Program Files\Windows Kits\10\build\WindowsDriver.common.targets(299,5): error MSB4062: The "ValidateNTTargetVersion" task could not be loaded from the assembly d:\Program Files\Windows Kits\10\build\bin\Microsoft.DriverKit.Build.Tasks.17.0.dll. Could not load file or assembly 'file:///d:\Program Files\Windows Kits\10\build\bin\Microsoft.DriverKit.Build.Tasks.17.0.dll' or one of its dependencies. The system cannot find the file specified. Confirm that the <UsingTask> declaration is correct, that the assembly and all its dependencies are available, and that the task contains a public class that implements Microsoft.Build.Framework.ITask.`

I'll repeat the steps with the Co and Ni EWDKs once they finish downloading.

samtertzakian, 2023-03-03:
Hi

Are you able to build using the VS IDE?

Other people are building using command line successfully.

I will investigate further. Thank you for the report.

77, 2023-03-05:
Sam, See my previous comment. I edited it with the details of my experiments.

samtertzakian, 2023-03-06:
Hi...

Thank you for your effort.

I am investigating this. I would like to see if we can make VS2022/226231 x64 work since msbuild works.

I will try to repro that and if I can I will see what I can do.

samtertzakian, 2023-03-06:
Hello....Ok, here is the status....

The error you are seeing with  VS2022/226231 is a known issue caused by a regression in WDK or VS2022 (not clear to me) and apparently there is an internal wokitem it is in the process of being corrected but I do not know the eventual outcome.

I was able to reproduce the error with VS2022/226231/x64. I was able to reproduce VS2022/226231/ARM64 works.

Finally, I was able to correct VS2022/226231/x64 by doing the following:

Before you start VS2022 from the 226231 EWDK command line, enter this command:

**set MSBuildUserExtensionsPath=%VSINSTALLDIR%\MsBuild**

Then, run VS2022. Then you should be able to build. 

Can you try that and let me know. I do not know if this will fix the other issues...but I think it fixes the latest version and probably the versions you want to build with.

Please let me know if that works for you...And thank you so much for the feedback. I am sure this will help others.

samtertzakian, 2023-03-06:
Unfortunately, that fix appears to not work for 220000 EWK. Please use 226231 instead.
I am not going to try the other EWDKs because the root problem appears to be a regression in VS2022.
The previous EWDKs work with VS2019.

So the two options are (I think):

1. Run **set MSBuildUserExtensionsPath=%VSINSTALLDIR%\MsBuild** command before running VS2022 (see above).
2. Use VS0219.

77, 2023-03-07:
Thanks Sam. To recap I did the following and I'm able to build both x64 and ARM64 from VS2022.

Mount EWDK_ni_release_svc_prod1_22621_220804-1759.iso on D:
`
D:
LaunchBuildEnv.cmd
cd BuildEnv
SetDevEnv.cmd
SetupBuildEnv.cmd
SetupVSEnv.cmd
set MSBuildUserExtensionsPath=%VSINSTALLDIR%\MsBuild
"c:\Program Files\Microsoft Visual Studio\2022\Enterprise\Common7\IDE\devenv.exe"
`

samtertzakian, 2023-03-07:
That is great to hear. 

Long term, when i get some time I will try to see if there is a way the projects can be modified so that that extra set does not need to be done. Also, I think VS2022 team is planning on fixing this issue, but it is not clear to me if that is true or when that would happen.

As for the issues with the older EWDK, I think it is best to use VS2019. If that does not work for you let me know.

Glad to hear you are unblocked. Please don't hesitate to create new issues should you find something else.

samtertzakian, 2023-03-28:
Note:

This issue is resolved in v1.1.132.

Starting with this version it is possible to build using both CO and NI EWDK with VS2019 and VS2022 **without** running this command:

`set MSBuildUserExtensionsPath=%VSINSTALLDIR%\MsBuild`

closed 2023-03-28 by samtertzakian