microsoft/DMF #248: WDF Verifier (Handle Tracking) trips over FILEOJECT de-reference without using the right tag

nefarius, 2023-12-04:
As per our e-mail conversation, a reminder that a bug check is triggered when WDF Verifier and Handle Tracking is turned on when the module `Dmf_NotifyUserWithRequestMultiple` is used. Fix as below:

```diff
 Dmf/Modules.Library/Dmf_NotifyUserWithRequestMultiple.c | 3 ++-
 1 file changed, 2 insertions(+), 1 deletion(-)

diff --git a/Dmf/Modules.Library/Dmf_NotifyUserWithRequestMultiple.c b/Dmf/Modules.Library/Dmf_NotifyUserWithRequestMultiple.c
index dc7dd75..32a8f9e 100644
--- a/Dmf/Modules.Library/Dmf_NotifyUserWithRequestMultiple.c
+++ b/Dmf/Modules.Library/Dmf_NotifyUserWithRequestMultiple.c
@@ -690,7 +690,8 @@ Return Value:
 
         // Dereference FileObject.
         //
-        WdfObjectDereference(fileObjectForDereference);
+        WdfObjectDereferenceWithTag(fileObjectForDereference, 
+									(VOID*)dmfModuleNotifyUserWithRequestMultiple);
     }
 
     // 3. Broadcast data to the Clients in the ListHead list.
```

Cheers!

samtertzakian, 2023-12-04:
Pending fix is here: https://github.com/microsoft/DMF/pull/249

samtertzakian, 2023-12-05:
Fixed in Release v1.1.142. https://github.com/microsoft/DMF/releases/tag/v1.1.142

closed 2023-12-05 by samtertzakian