microsoft/DMF #252: Dmf_VirtualHidKeyboard work in User-mode?

UMU618, 2024-03-04:
https://github.com/microsoft/DMF/blob/08132eb4895be03ea673791a755b0dbb7d1d8e11/Dmf/Modules.Library/Dmf_VirtualHidKeyboard.h#L16

https://github.com/microsoft/DMF/blob/08132eb4895be03ea673791a755b0dbb7d1d8e11/Dmf/Modules.Library/Dmf_VirtualHidKeyboard.c#L18

I think the Comments is wrong.

samtertzakian, 2024-03-07:
Yes, the issue is that the Child Module DMF_VirtualHidDeviceVhf is only supported in Kernel mode. However, it is in the plan to make it User-mode. (When we wrote the Module, VHF was only supported in Kernel-mode but now is supported in User-mode.) We will make the update and let you know. I will leave this issue open until then.

paulpv, 2024-11-13:
@samtertzakian Is this working now?

I am a little confused by your comment:
1. > Yes, the issue is that the Child Module DMF_VirtualHidDeviceVhf is only supported in Kernel mode. However, it is in the plan to make it User-mode.
1. > (When we wrote the Module, VHF was only supported in Kernel-mode but now is supported in User-mode.)

#2, that VHF is now supported in both Kernel and User mode, seems to contradict #1, that DMF_VirtualHidDeviceVhf is only supported in Kernel mode and is planned to support User mode one day.

Would it be possible for there to be a sample of proper usage of a `Dmf_VirtualHidKeyboard` in both Kernel-mode and User-mode?