microsoft/DMF #260: Documentation needs to include adding `<ClCompile ...><Wpp* ...></ClCompile>` to VS project?

paulpv, 2024-11-09:
I wanted to learn how to create a DMF project, so I:
1. Cloned DMF
1. Built the DMF solution successfully
1. In Visual Studio 2022 [Community] added a new standard template KMDF project and removed some its boilerplate code
1. Manually integrated `Dmf.props` into my KMDF DMF project following the [IMHO important but very buried] instructions at:  
   https://github.com/Microsoft/DMF/blob/master/Dmf/Documentation/Driver%20Module%20Framework.md#simplifying-compilation-and-linking-with-dmf
1. Added some basic hello world DMF code from:  
   https://github.com/Microsoft/DMF/blob/master/Dmf/Documentation/Driver%20Module%20Framework.md#using-dmf-in-a-driver-that-does-not-have-a-deviceadd-callback  
   Roughly...
```
#include "DmfModules.Library.h"
DRIVER_INITIALIZE DriverEntry;
EVT_WDF_DRIVER_DEVICE_ADD StorageFwUpdateDeviceAdd;
EVT_WDF_OBJECT_CONTEXT_CLEANUP StorageFwUpdateDriverContextCleanup;
EVT_DMF_DEVICE_MODULES_ADD DmfDeviceModulesAdd;

/*WPP_INIT_TRACING(); (This comment is necessary for WPP Scanner.)*/
#pragma code_seg("INIT")
DMF_DEFAULT_DRIV<RENTRY(DriverEntry, 
                        StorageFwUpdateDriverContextCleanup,
                        StorageFwUpdateDeviceAdd)
#pragma code_seg()

#pragma code_seg("PAGED")
DMF_DEFAULT_DRIVERCLEANUP(StorageFwUpdateDriverContextCleanup)
DMF_DEFAULT_DEVICEADD(StorageFwUpdateDeviceAdd,
                      DmfDeviceModulesAdd)
#pragma code_seg()
```
6. Built the project and got this errors saying
```
Severity	Code	Description	Project	File	Line	Suppression State	Details
Error	C4013	'WPP_CLEANUP' undefined; assuming extern returning int	KMDF WDF DMF VHF KMClone	DmfInterface.c	9		
Error	C4013	'WPP_INIT_TRACING' undefined; assuming extern returning int	KMDF WDF DMF VHF KMClone	DmfInterface.c	9		
```

I diffed my project with the DmfSamples and found that the DmfSamples had this:
```
#include "Trace.h"
#include "DmfInterface.tmh"
```

I had my own `Trace.h` from the template.
But what was this "DmfInterface.tmh"?

It took me about half a day to this nugget of goodness in all of the DmfSample projects:
```
  <ItemGroup Label="WrappedTaskItems">
    <ClCompile Include="DmfInterface.c">
      <WppEnabled>true</WppEnabled>
      <WppKernelMode>true</WppKernelMode>
      <WppTraceFunction>TraceEvents(LEVEL,FLAGS,MSG,...)</WppTraceFunction>
      <WppGenerateUsingTemplateFile>{km-WdfDefault.tpl}*.tmh</WppGenerateUsingTemplateFile>
    </ClCompile>
  </ItemGroup>
```

I also learned something new: WPP  
* https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/tracewpp-task
* https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/wpp-software-tracing

I think this `ClCompile Wpp` section needs to be in the documentation at:  
https://github.com/Microsoft/DMF/blob/master/Dmf/Documentation/Driver%20Module%20Framework.md

samtertzakian, 2024-11-14:
Hmm.... This is strange. For us, all the sample drivers compile. Let me investigate this issue. It is not clear to me why it was not working for you. 
Thank you for the information.

paulpv, 2024-11-15:
> Hmm.... This is strange. For us, all the sample drivers compile. Let me investigate this issue. It is not clear to me why it was not working for you. Thank you for the information.

All of the sample drivers do work fine for me.

It is when I create my own new KMDF [DMF] driver project from scratch that I have to add that undocumented config to the project file.

nefarius, 2024-11-17:
Maybe this is an issue with the WDK templates rather than DMF? In none of my DMF projects do I need to add this snippet, it should be compatible with WPP in project skeletons out of the box. Maybe share your complete demo project, if you can (anonymised ofc.).